adamreeve/npTDMS #221: iterating over goups, channels with items()

jmitrevs, 2020-11-09:
Given that you can iterate, for example, with
```python
for group_name in tdms_file:
    ...
```

```python
for channel_name in group:
    ...
```
I think it would be useful to also allow:
```python
for channel_name, channel in group.items():
    ...
```
This is of course not a key feature, but I think it would be nice.

adamreeve, 2020-11-15:
You can get something close to this with the `groups` and `channels` methods:
```python
for group in tdms_file.groups():
    group_name = group.name
    ...

for channel in group.channels():
    channel_name = channel.name
    ...
```

But I agree, this would be nice given that the `TdmsFile` and `TdmsGroup` classes already act like dictionaries in other ways.

adamreeve, 2021-02-20:
I had a look at implementing this but have changed my mind and think the `items` approach is less clear than just using the `groups` and `channels` methods, and doesn't really add any benefit. Feel free to rely if you think I'm wrong.

closed 2021-02-20 by adamreeve