adamreeve/npTDMS #226: Can't open file

jialonso, 2021-01-12:
Hi!

I'm having some problem opening .tdms files. Files can be open with the excel plugin, but the library throws an exception.

When I run:
```python
import nptdms
nptdms.TdmsFile.open('POC_MultisamplingRate.tdms')
```
I'm getting: `UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 12: invalid start byte`.

The attached .zip file contains 3 files. Two of them can't be open (POC_MultisamplingRate.tdms and POC_ThreeGroup.tdms) while the other one (POC_OneGroup.tdms) works well.
[TDMS_files.zip](https://github.com/adamreeve/npTDMS/files/5800819/TDMS_files.zip)

Thank you in advance!

adamreeve, 2021-01-16:
Hi, thanks for the bug report.

It looks like the problem is that these files have segments without the `kTocDAQmxRawData` flag set, but containing channels with DAQmx raw data, indicated by the `69 12 00 00` raw data index header at the beginning of the channel metadata. Currently npTDMS only looks at the segment flag to determine whether channel data is DAQmx data, but it looks like it must be valid to have DAQmx channel data when this flag isn't set.

adamreeve, 2021-01-16:
Looking into this a bit more, it's not really clear yet what is going on as these channels don't seem to fit the structure of a normal DAQmx raw data channel and just treating them as a DAQmx channel causes a different crash. Eg. the channel data type is usually set to the DAQmx raw data type for a DAQmx channel, but in the POC_ThreeGroup file the `/'FullRate'/'TimeRef'` channel has a TimeStamp data type but a DAQmx raw data index header.

Ummon, 2021-02-03:
Hello,
I think that _DAQmx_ channels can be mixed with _non-DAQmx_ channels within a segment. The flag `ContainsRawData` in the ToC segment is just a clue that it may contain some _DAQmx_ data channels (not sure).

The `Raw data index` indicates if the channel contains _DAQmx_ data with the value `69 12 00 00` (Format Changing scaler) or `69 13 00 00` (Digital Line scaler).

The file `POC_MultisamplingRate.tdms` contains only _DAQmx_ channels and only one segment. Have you tried to set the flag `ContainsRawData` to 1 in the file (offset 4, bit 7) then rerun _npTDMS_?

Reference: https://www.ni.com/en-gb/support/documentation/supplemental/07/tdms-file-format-internal-structure.html

adamreeve, 2021-02-07:
I've made a bit more progress understanding what's going on here but do not yet have a working solution. It looks like the segment does not actually have DAQmx raw data for the data itself, but the object metadata is in the format used by DAQ mx data. One other thing that's causing problems is that the segment has interleaved data but the channels in the segment are of different lengths, so it looks like the groups may be interleaved separately.

adamreeve, 2021-02-08:
It looks like the file data is in DAQmx format despite the segment not having the DAQmx data flag, but the data types are set to TimeStamp and int64 whereas usually I'd expect these to just be the DAQmx data type. The other thing causing a problem is that there are three separate raw data buffers with different data lengths in each, whereas currently the npTDMS code expects all DAQmx channels to be the same length.

jialonso, 2021-02-10:
Thank you @adamreeve for your comments. In fact, the main goal of these TDMS files is to be able to provide data at multiple sample rates.

I have been using TDMS files for a while, but thank to your library it was really simple to open them and I have never get into the internal details of the format. In any case, if there is something I can help you, please tell me.

adamreeve, 2021-03-04:
Hi @jialonso, I've now merged a fix for this in #231 with further explanation there of the changes needed. Thanks for your examples, the POC_ThreeGroup.tdms file in particular made it quite clear what the structure of the data was.

closed 2021-03-12 by adamreeve