adamreeve/npTDMS #228: [nptdms.scaling WARNING] Unsupported scale type: Strain

Hritik-Raj, 2021-02-20:
Hi!

I am just starting to work with TDMS data files, and I must say, this library is very well designed. I have been facing some issues with scaling however, and it seems that npTDMS is not allowing me to use the Strain scale.

This script halts as soon as one of the channels uses a strain scale

import numpy as np
from nptdms import TdmsFile as td
import pandas as pd
import datetime
tdms_file = td.read('/Users/aj/Desktop/FTA_CTA_Trains_37052.tdms')
num_groups = len(tdms_file.groups())
group = tdms_file['FTA_CTA_Train']
for item in group:
    print(group[item].read_data(scaled=True))

This is the error that it throws -

[nptdms.scaling WARNING] Unsupported scale type: Strain
Traceback (most recent call last):
  File "version1.py", line 9, in <module>
    print(group[item].read_data(scaled=True))
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/nptdms/tdms.py", line 590, in read_data
    return self._scale_data(raw_data)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/nptdms/tdms.py", line 743, in _scale_data
    raise ValueError("Missing scaling information for DAQmx data")
ValueError: Missing scaling information for DAQmx data

adamreeve, 2021-02-20:
I've only added support for new scalings when they have been requested as there isn't documentation on how NI implements each scaling, so it can take some time to get the implementation correct. Are you able to provide an example file that uses strain scaling?

Based on the documentation at https://zone.ni.com/reference/en-XX/help/371361R-01/glang/tdms_create_scalinginfo/, it looks like the scaling information should have properties for configuration, Poisson ratio, gage resistance, lead wire resistance, initial bridge voltage, gage factor and voltage excitation, so it is just a matter of working out how to compute the strain from the measured voltage signal given these properties. Is this something you could help with?

Hritik-Raj, 2021-02-20:
Hi,

Thank you for your quick response. While I am not familiar enough with Python to implement this, I can certainly try to help. How should I provide the example file? Github does not let me upload TDMS format

adamreeve, 2021-02-20:
I believe GitHub should let you upload a zip file. Otherwise my email address can be found in the setup.py file if you'd like to email me an example: https://github.com/adamreeve/npTDMS/blob/master/setup.py

Implementing the scaling in Python isn't really the tricky part, it's working out the formula for computing the strain given the input voltage, and my electronics/instrumentation knowledge is quite rusty now.

Hritik-Raj, 2021-02-20:
I emailed you with the files!

adamreeve, 2021-02-23:
Thanks Hritik, your TDMS files have strain scaling properties like:
```python
 ('NI_Scale[2]_Scale_Type', 'Strain'),
 ('NI_Scale[2]_Strain_Configuration', 10183),
 ('NI_Scale[2]_Strain_Poisson_Ratio', 0.3),
 ('NI_Scale[2]_Strain_Gage_Resistance', 350.0),
 ('NI_Scale[2]_Strain_Lead_Wire_Resistance', 0.0),
 ('NI_Scale[2]_Strain_Initial_Bridge_Voltage', 0.0),
 ('NI_Scale[2]_Strain_Gage_Factor', 2.1),
 ('NI_Scale[2]_Strain_Bridge_Shunt_Calibration_Gain_Adjustment', 1.0),
 ('NI_Scale[2]_Strain_Voltage_Excitation', 2.5),
```

NI have published a helpful article that explains different strain gauge configurations at at https://www.ni.com/en-in/innovations/white-papers/07/measuring-strain-with-strain-gages.html

Your files use a full Wheatstone bridge with configuration type I and in that case the equation for strain in terms of the measured voltage simplifies to &epsilon; = - V<sub>0</sub> / (GF * V<sub>ex</sub>), ignoring lead wire resistance, initial bridge voltage and shunt calibration gain adjustment.

I can implement a scaling that just handles full bridge type I for now.

Hritik-Raj, 2021-02-23:
Hi Adam,

Thanks for your help with this, that would be great!

adamreeve, 2021-02-24:
This is now fixed on the master branch

PR #230 (2021-02-24): Implement strain scaling for full bridge type I
Fixes #228
closed 2021-02-24 by adamreeve