adamreeve/npTDMS #239: UnicodeDecodeError at start byte

admond-lee-mu, 2021-05-08:
Hi, 

In tdms file [Sample.zip](https://github.com/adamreeve/npTDMS/files/6445297/Sample.zip) which opens totally fine with Lab View throws exception:

> UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 10: invalid start byte

As far as I have tried, I guess this is because of the [limitation]([Sample.zip](https://github.com/adamreeve/npTDMS/files/6445297/Sample.zip)) of this package.
Do you have any idea or solution?

Thanks in advance.

Note. Text is enciphered on purpose.

adamreeve, 2021-05-08:
Hi @admond-lee-mu, it looks like the issue is that your file contains string data in an interleaved data segment, which npTDMS doesn't handle correctly. Having interleaved string data doesn't make much sense to me as string values can vary in size, but it looks like these segments only contain a single channel so should be able to be treated as if they're non-interleaved.

closed 2021-05-09 by adamreeve

adamreeve, 2021-05-09:
This has been fixed in version 1.3.0