adamreeve/npTDMS #245: AttributeError: type object 'TdmsFile' has no attribute 'object'

claypilat, 2021-06-18:
Hi, 

When trying to pull properties from a TDMS file I am able to pull the name and value as noted in the documentation like the following: 

```
tdms_file = TdmsFile.read("file.tdms")
for name, value in tdms_file.properties.items():
    print("{0}: {1}".format(name, value))

```

However, I know there are additional properties generated. Looking at some code examples I should be able to pull them using something such as 

```
def showRootProps(oTdmsFile):
    # method .properties returns a dictionary
    root_props = oTdmsFile.object().properties.items()
    print ('File properties:')
    for key,value in root_props:
        print ("\t{0} : {1} ".format(key,value))

```

I've used variations to try read the object properties but I keep getting thrown the error
"TdmsFile' object has no attribute 'object'. Am I trying to access this incorrectly?

adamreeve, 2021-06-18:
Hi @claypilat, the `TdmsFile.object()` method used to work with older versions of npTDMS to return the root object, but is no longer supported. The `properties` attribute used in your first example should include all properties of the file.

What makes you say there are additional properties generated? There may also be properties associated with the groups and channels in the file, which can also be accessed with the properties attribute on these objects, eg:
```python
tdms_file["group name"].properties
tdms_file["group name"]["channel name"].properties
```

claypilat, 2021-06-21:
You legend. Apologies I was reading through older versions of documentation. The properties I'm looking for are hidden in the channels. I've written a function to loop through the channels and groups and pull those property items. Thanks for your prompt response.

closed 2021-06-21 by claypilat