adamreeve/npTDMS #267: Reading strain gauge files

BockiBock, 2022-02-28:
Hey community,

thanks for creating such a helpful library. I use this often to work on tdms-files for voltage measurements.
To read the files, I use this kind of code: 
```

from nptdms import TdmsFile
tdms_file = TdmsFile.read("C:/Users/Desktop/DDMS.tdms")
group = tdms_file.groups() 
channel = group[0].channels()

Q1 = channel[0][:]

```
When I try to read strain gauge files I get the following error:

`Exception: Strain gauge configuration 10188 is not supported`

Unfortunately, I could not find any solution so far. Did anyone else had a similar problem before?
Thanks for your help in advance.

adamreeve, 2022-03-01:
Linking related discussion: https://github.com/adamreeve/npTDMS/discussions/266

Currently only full bridge type I is implemented, support for half-bridge type I (10188) requires implementing the computation of strain from the measured voltage.

BockiBock, 2022-03-03:
Hey Adam,

it's me again.
I fixed the problem by adding the following code in the function def scale(
self, data): (line 197 of scaling)

        if self.configuration == StrainScaling.HALF_BRIDGE_1:
            if self.initial_bridge_voltage != 0.0:
                voltage_diff = data - self.initial_bridge_voltage
            else:
                voltage_diff = data
                ((-4*voltage_diff) / (self.gage_factor*((1+self
.poisson_ratio)-2*(voltage_diff-1))))*(1+(self.lead_wire_resistance/self
.gage_resistance))
            return voltage_diff * (-self.gain_adjustment /
(self.voltage_excitation
* self.gage_factor))

The results also look reasonable, so I guess that does the trick.

Thanks again for the help and maybe you can update your library.
Have a good day.

Kind regards
Tom

BockiBock, 2022-03-03:
And a quick update, since I only looked at the run of the curves, but not
the amount of the single values...
   def scale(self, data):
        """ Convert voltage data to strain
        """
        if self.configuration == StrainScaling.FULL_BRIDGE_1:
            if self.initial_bridge_voltage != 0.0:
                voltage_diff = data - self.initial_bridge_voltage
            else:
                voltage_diff = data
            return voltage_diff * (-self.gain_adjustment /
(self.voltage_excitation
* self.gage_factor))

        if self.configuration == StrainScaling.HALF_BRIDGE_1:
            if self.initial_bridge_voltage != 0.0:
                voltage_diff = data - self.initial_bridge_voltage
            else:
                voltage_diff = data
            return voltage_diff * (((-4*self.gain_adjustment) / ((self
.voltage_excitation*self.gage_factor)*((1+self.poisson_ratio)-2*(self
.gain_adjustment-1))))*(1+(self.lead_wire_resistance/self.gage_resistance)))

        raise Exception("Strain gauge configuration %d is not supported" %
self.configuration)

    FULL_BRIDGE_1 = 10183
    FULL_BRIDGE_2 = 10184
    FULL_BRIDGE_3 = 10185
    HALF_BRIDGE_1 = 10188
    HALF_BRIDGE_2 = 10189
    QUARTER_BRIDGE_1 = 10271
    QUARTER_BRIDGE_2 = 10272

Now it should do the job, but please let me know if you find any other
mistake.

sorry for the spam and thanks again for the help

adamreeve, 2022-03-03:
Awesome, thanks Tom, I'll make sure to integrate this into the library.

closed 2022-03-10 by adamreeve

adamreeve, 2022-03-10:
Hi @BockiBock, I've now merged this to master and added the other strain gauge configurations as well. The equation I ended up with is slightly different to yours and more closely matches the output from LabView. The `gain_adjustment` only multiplies the whole strain value, and I've included the extra `2 (Vo / Vex) (1 - ν)` term that makes the equation non-linear.

https://github.com/adamreeve/npTDMS/blob/b4396c7d97f1ffb718bd38e6dd3da7f785991e91/nptdms/scaling.py#L243-L249

PR #268 (2022-03-10): Support remaining strain gauge configurations
Fixes #267