adamreeve/npTDMS #279: No easy way to write the properties of existing file

mLichtenheld, 2022-07-15:
hi,
so this is more a feature request then an actual issue.
I wanted to edit the tdms file-properties of many tdms files.

I cant find a way without rewriting all those data.
Am I missing something?

nevertheless, great work!
its a helpfull and easy to use module 👍

adamreeve, 2022-07-17:
Hi. The way the TDMS format works, you can modify properties by appending a segment with the new property values and any existing property values will be overridden. So you should be able to modify properties by opening a `TdmsWriter` in append mode and adding a segment that only contains metadata.

One issue I discovered recently is that you need to ensure the version used matches the existing file version otherwise some readers will raise errors, although this will probably always be 4713 for most files. This could possibly be automated to make this a bit simpler. It's also slightly awkward if modifying channel properties as npTDMS currently expects a `ChannelObject` to have data, so you'd need to create an empty array of the correct data type. This can probably be improved too so you can use channel objects without data.

For example, this code will append a segment with new properties for the file and a channel:

```python
with TdmsFile.open(file_path) as f:
    version = f.tdms_version

empty_data = np.array([], dtype=np.float64)

with TdmsWriter(file_path, mode='a', version=version) as f:
    root = RootObject(properties={'file_property_name': 'property_value'})
    channel = ChannelObject(
        'group_name', 'channel_name', empty_data,
        properties={'channel_property_name': 'property_value'})
    f.write_segment([root, channel])
```

mLichtenheld, 2022-07-19:
thank you very much!
I wasnt aware I could append to file with the tdmsWriter.
But your example made much sense and it works just fine!
great

closed 2022-07-19 by mLichtenheld