adamreeve/npTDMS #280: Read metadata from index file only

Joe-Heffer-Shef, 2022-08-23:
Hi,

Is it possible to read the metadata for a TDMS file using only the `.tdms_index` file?

I have a use-case where I want to avoid downloading a full `.tdms` file just to read the metadata.

Eifi1, 2022-08-23:
Hello Joe,

yes, with version 1.6.0 it is possible to read metadata.
You can traverse through the file and their objects by using `nptdms.TdmsFile.open(filepath)` without trying to read the data.

Best regards

Joe-Heffer-Shef, 2022-08-24:
> yes, with version 1.6.0 it is possible to read metadata. You can traverse through the file and their objects by using `nptdms.TdmsFile.open(filepath)` without trying to read the data.

Thanks for your reply @Eifi1. I understand how to use `nptdms.TdmsFile.read_metadata`.

Is it possible to read the metadata by reading *only* the `.tdms_index` file (*without* the `.tdms` file, which has a much larger file size.)

adamreeve, 2022-08-24:
Yes, the ability to read metadata from only a `.tdms_index` file was added in version 1.6.0, specifically this pull request: https://github.com/adamreeve/npTDMS/pull/278

The code example above is for reading a `.tdms_index` file, you can simply pass the path to an index file to the `open` method rather than the path to the `.tdms` file.

Have you tried this and found that it didn't work?

I realise now that there wasn't any documentation added for this functionality, which should be addressed.

Eifi1, 2022-08-24:
> Is it possible to read the metadata by reading _only_ the `.tdms_index` file (_without_ the `.tdms` file, which has a much larger file size.)

Please find the snippet below, that worked for me. You will get an error, when you try to `channel[:]` or `channel.read_data()`, but metadata can be read.

    import nptdms
    path_to_index_file = ""
    file = nptdms.TdmsFile.read_metadata(path_to_index_file)
    print(file.properties)`
    for group in file.groups():
        print(group.properties)
        for channel in group.channels():
            print(channel.properties)
            print(channel.name)

You can even get the segments positions a specific channel consists of from the `.tdms_index` file.
With this you can stream not the whole `.tdms` but only a small portion. Works better when using defragmented files where each channel is represented by a single tdms segment.

    import nptdms
    path_to_index_file = "M2074.tdms_index"
    with nptdms.TdmsFile.open(path_to_index_file) as f:
        for channel in f.groups()[0].channels():
        positions = []
        for segment in f._reader._segments:
            if channel.path in [o.path for o in segment.ordered_objects]:
                positions.append((segment.position, segment.next_segment_pos))
        print(positions)

Joe-Heffer-Shef, 2022-09-01:
I tried this like so:

```python
import logging
import argparse
from pathlib import Path

import nptdms

DESCRIPTION = """
https://github.com/adamreeve/npTDMS/issues/280
"""

logger = logging.getLogger(__name__)

def get_args() -> argparse.Namespace:
    """
    Command line arguments
    """
    parser = argparse.ArgumentParser(description=DESCRIPTION)

    parser.add_argument('-i', '--index', help='TDMS index file path', type=Path, required=True)

    return parser.parse_args()

def main():
    args = get_args()
    logging.basicConfig(level=logging.INFO)

    with args.index.open('rb') as file:
        tdms_file = nptdms.TdmsFile.read_metadata(file)
        logger.info(f"Loaded '{file.name}' ({tdms_file})")

if __name__ == '__main__':
    main()
```

this is the result

```
>python . -i "AI4761R_2_CFRP4.7 GapVar Al6_C1_4_1_data_201207_112550.tdms_index"
Traceback (most recent call last):
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\my_username\my_project\scripts\tdms-index\__main__.py", line 35, in <module>
    main()
  File "C:\Users\my_username\my_project\scripts\tdms-index\__main__.py", line 30, in main
    tdms_file = nptdms.TdmsFile.read_metadata(file)
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 100, in read_metadata
    return TdmsFile(file, raw_timestamps=raw_timestamps, read_metadata_only=True)
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 130, in __init__
    self._read_file(self._reader, read_metadata_only, keep_open)
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 226, in _read_file
    tdms_reader.read_metadata(require_segment_indexes=keep_open)
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 87, in read_metadata
    segment, properties = self._read_segment_metadata(
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 235, in _read_segment_metadata
    (position, toc_mask, data_position, next_segment_pos, segment_incomplete) = self._read_lead_in(
  File "C:\Users\my_username\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 253, in _read_lead_in
    raise ValueError(
ValueError: Segment does not start with b'TDSm', but with b'TDSh'
```

Joe-Heffer-Shef, 2022-09-01:
I also tried the script from https://github.com/adamreeve/npTDMS/issues/280#issuecomment-1225471970

```python
import nptdms

path_to_index_file = "Low Speed task1_220703_131300.tdms_index"

with nptdms.TdmsFile.open(path_to_index_file) as f:
    for channel in f.groups()[0].channels():
        positions = []
    for segment in f._reader._segments:
        if channel.path in [o.path for o in segment.ordered_objects]:
            positions.append((segment.position, segment.next_segment_pos))
    print(positions)
```

and I got:

```
$ python test.py
Traceback (most recent call last):
  File "C:\Users\me\my_project\scripts\tdms-index\test.py", line 4, in <module>
    with nptdms.TdmsFile.open(path_to_index_file) as f:
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 87, in open
    return TdmsFile(
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 130, in __init__
    self._read_file(self._reader, read_metadata_only, keep_open)
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\tdms.py", line 226, in _read_file
    tdms_reader.read_metadata(require_segment_indexes=keep_open)
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 87, in read_metadata
    segment, properties = self._read_segment_metadata(
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 235, in _read_segment_metadata
    (position, toc_mask, data_position, next_segment_pos, segment_incomplete) = self._read_lead_in(
  File "C:\Users\me\Miniconda3\envs\my_project\lib\site-packages\nptdms\reader.py", line 253, in _read_lead_in
    raise ValueError(
ValueError: Segment does not start with b'TDSm', but with b'TDSh'
```

Eifi1, 2022-09-01:
> $ python -c "import nptdms; print(nptdms.__version__)"
> 1.4.0
> $ python --version
> Python 3.9.7

The ability to read index files only was introduced in version 1.6.0. please update nptdms and tell us if the problem persists.

Joe-Heffer-Shef, 2022-09-01:
D'oh! Thanks very much :)

Yes, updating to version 1.6 fixes this issue.

```
$ python -c "import nptdms; print(nptdms.__version__)"
1.6.0
```

For the first script https://github.com/adamreeve/npTDMS/issues/280#issuecomment-1234270799

```bash
$ python . -i "Low Speed task1_220703_131300.tdms_index"
INFO:__main__:<nptdms.tdms.TdmsFile object at 0x000002A34A227FD0>
```
and the second script https://github.com/adamreeve/npTDMS/issues/280#issuecomment-1234309274
```bash
$ python test.py
[(0, 8192), (8192, 16232448), (16232448, 16236560)]
```

closed 2022-09-01 by adamreeve