adamreeve/npTDMS #281: Tdms file issue

rabzgzz, 2022-09-01:
Hello, 

I'm trying to read a series of tdms files using the "TdmsFile.read()" method in python and I'm getting the following error:

Exception has occurred: ValueError
Segment does not start with b'TDSh', but with b'\x00\x00\x00\x00'

I checked and the files are all valid Tdms files as they all start with the TDSm characters. Could you give me any advice on how to address this issue? 

Thank you in advance.

adamreeve, 2022-09-02:
Hi, the error says that it's expecting 'TDSh' rather than 'TDSm', so it looks like there's an issue reading the `.tdms_index` files. Does the error go away if you remove or rename the `.tdms_index` files?

rabzgzz, 2022-09-02:
Hello Adam thank you for your reply! I just tried both things but the error keeps appearing.

adamreeve, 2022-09-02:
Okay, I'm not sure what's going on then. Are you able to provide an example file that reproduces the error?

rabzgzz, 2022-09-02:
Sure, the file exceeds the size limit here (even when compressed as .zip). I could send it to you by mail if you agree.

adamreeve, 2022-09-02:
Yep sure, I'm adreeve@gmail.com, thanks

adamreeve, 2022-09-08:
This issue appears to be related to reading the data from a cloud share, as reading the files locally didn't reproduce the error. It's not clear exactly what's going wrong but this can be worked around by copying the files locally, so I'm going to close this issue.

closed 2022-09-08 by adamreeve