adamreeve/npTDMS #283: Reading TDMs Files

dinesh-m2, 2022-09-29:
I'm not able to read a TDMs file. 
ValueError: Segment does not start with b'TDSm', but with b'LDsf'

I've tried all the methods. The solution mention in [#123 ](https://github.com/adamreeve/npTDMS/issues/123) doesn't work in my case. Can someone please help me out.
https://user-images.githubusercontent.com/88016397/193087645-189d2318-2c60-4842-856c-d39dbd551712.jpg

Eifi1, 2022-09-29:
Hello, could you provide an example file.

Could you furthermore make a screenshot from the file opened with a simple text editor? To make sure the file on the disk is valid.

And at last the first bytes, that appear, when reading with python with the following code snippet from the same location you tried to read with TdmsFile(filepath)?

    with open(file path, 'rb') as f:
        print(f.read(10))

fostermaier, 2023-04-06:
I stumbled upon this issue. If anyone sees

    ValueError: Segment does not start with b'TDSh', but with b'vers'

specifically referencing `vers`, the file has not been pulled correctly by LFS.

    with open(file, 'r') as f:
        print(f.read(1000000))
    
    >> version https://git-lfs.github.com/spec/v1
    >> oid sha256:274f1611b31817a56d406e8638ab15988872003e22162cb7e412
    >> size 57193

other issues might be similar.

thomasnelissen, 2025-01-30:
I am getting the same error here:
ValueError: Segment does not start with b'TDSm', but with b'LDsf'

My tdms file starts with a 'LDsf' byte:

[image]

adamreeve, 2025-02-02:
> My tdms file starts with a 'LDsf' byte. Any clues?

It looks like your "tdms file" is not actually a tdms file but is some other format. What makes you think it's a TDMS file?