adamreeve/npTDMS #296: Problem with reading Sp

dulacskamty, 2023-01-30:


adamreeve, 2023-01-30:
Hi, my intention is that this should support any file-like object, but this relies on having a common interface for what should be considered a file. The `readinto` method is defined on `io.RawIOBase` or `io.BufferedIOBase` so any class inheriting from either of those or implementing the same interface should work.

According to the tempfile docs, `SpooledTemporaryFile` was changed in Python version 3.11 to fully implement the `io.BufferedIOBase` interface. Is upgrading to Python 3.11 an option for you?

dulacskamty, 2023-01-31:
Thank you for the quick answer and for looking into the problem. You are right, reading SpooledTemporaryFile works well with Python version 3.11, so I close this issue.

closed 2023-01-31 by dulacskamty