adamreeve/npTDMS #297: Error in loading image files

Nablaaa, 2023-02-02:
I wanted to load a tdms file which I acquired with a microscope (1024x1024 px). This file consist of 5 images, so  the total amount of entries in the list is 5x1024x1024. 

The npTDMS script:

```
from nptdms import TdmsFile
tdms_file = TdmsFile.read("test_0/cam1/event1.tdms")
group = tdms_file['img']
channel = group["cam1"]
data = channel[:]
print(len(data))
```

gave me a slightly larger number, which contained all information from the 5 images, plus a couple of additional pixels.

I found out that this comes from the "tdms_segment.py" file in line 296, where the chunk_remainder != 0
here it tries to  use the function "_compute_final_chunk_lengths", which does not work in the end.

I just skipped the problem by setting

if chunk_remainder == 0:
            self.num_chunks = int(total_data_size // data_size)
else:
    self.num_chunks = int(total_data_size // data_size)

or basically by skipping the if/else condition and just using self.num_chunks = int(total_data_size // data_size).

This works for me and creates the desired output (a list of size 5x1024x1024) which i can reshape to 5 images of size 1024x1024.

Now I wanted to know if there is another way of reading image .tdms files with npTDMS? Also I wanted to know if this is a bug or if this is made on purpose for other kind of data (like tables etc.)?

Best wishes,
Eric

adamreeve, 2023-03-12:
Hi, I'm not sure what's going on here, I haven't worked with image data in TDMS files before so possibly this is a peculiarity specific to that scenario, but the TDMS format doesn't have any built in support for 2D image data as far as I know, and there's no special way to read image data with npTDMS.

I've seen files with normal 1D data that have a data size that isn't a multiple of the chunk size, and had assumed this might be caused by a crash that lead to the full chunk not being written. So npTDMS will read this data in case it is useful, but will log that warning so users know there's something not quite right.

Nablaaa, 2023-03-12:
Alright thank you for your reply and thanks for logging the warning.

Have a nice day!

closed 2023-03-12 by Nablaaa