adamreeve/npTDMS #313: How to ignore `warning` in read function?

OUCyf, 2023-10-16:
Gteetingd,
super great package!! I'm Fu. Recently, I use `nptdms` to read some `tdms` file, and i want to ignore those warnings, for example:

```
import warnings
from nptdms import TdmsFile

with warnings.catch_warnings():
    warnings.simplefilter("ignore")
    tf = TdmsFile.read(file_tdms)
```

Although i used `warnings.simplefilter("ignore")`, and will get:

```
[nptdms.tdms_segment WARNING] Data size 291840000 is not a multiple of the chunk size 1050624. Will attempt to read last chunk
WARNING:nptdms.tdms_segment:Data size 291840000 is not a multiple of the chunk size 1050624. Will attempt to read last chunk
```

So how to remove those warnings?

Best,
Fu

adamreeve, 2023-10-16:
Hi Fu, these warnings aren't from the warnings library but are using the Python logging library. You can configure the logging level to something higher than warnings to hide them, eg:

```python
import logging
from nptdms.log import log_manager
log_manager.set_level(logging.ERROR)
```

adamreeve, 2023-11-07:
Closing this as answered, feel free to reopen if this doesn't solve your issue

closed 2023-11-07 by adamreeve

OUCyf, 2023-11-07:
Oh yep, forget to reply that it works well!
Thanks so much!

Best,
Fu