adamreeve/npTDMS #329: 4 tests fail

yurivict, 2024-04-05:
```
===================================================================================== warnings summary ======================================================================================
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[2-0.0-expected_data0]
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[2-100.0-expected_data1]
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[3-0.0-expected_data2]
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[3-100.0-expected_data3]
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[4-0.0-expected_data4]
nptdms/test/test_scaling.py::test_thermistor_scaling_with_voltage_excitation[4-100.0-expected_data5]
  /usr/ports/misc/py-npTDMS/work-py39/npTDMS-1.9.0/nptdms/scaling.py:398: RuntimeWarning: invalid value encountered in reciprocal
    r_t = self.r1_reference_resistance * np.reciprocal(self.excitation_value * np.reciprocal(data) - 1.0)

nptdms/test/test_scaling.py: 12 warnings
  /usr/ports/misc/py-npTDMS/work-py39/npTDMS-1.9.0/nptdms/scaling.py:406: RuntimeWarning: invalid value encountered in reciprocal
    return np.reciprocal(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html

------------------------------------------------------------------------------------------------------------- benchmark: 2 tests ------------------------------------------------------------------------------------------------------------
Name (time in us)                                                   Min                    Max                Mean              StdDev              Median                IQR            Outliers  OPS (Kops/s)            Rounds  Iterations
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_thermocouple_scaling_voltage_to_temperature_benchmark     308.8300 (1.0)      26,576.0350 (1.23)     408.4210 (1.0)      582.6994 (1.05)     388.3135 (1.0)      15.2690 (1.0)         20;92        2.4485 (1.0)        2072           1
test_thermocouple_scaling_temperature_to_voltage_benchmark     481.6630 (1.56)     21,656.9110 (1.0)      587.4312 (1.44)     554.6138 (1.0)      568.7380 (1.46)     18.4478 (1.21)         3;92        1.7023 (0.70)       1457           1
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

---------------------------------------------------------------------------------- benchmark 'read-all-channel': 6 tests ----------------------------------------------------------------------------------
Name (time in ms)                            Min                 Max               Mean             StdDev             Median               IQR            Outliers       OPS            Rounds  Iterations
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_stream_interleaved_data_channel      3.8703 (1.0)        7.4332 (1.0)       4.2199 (1.0)       0.3537 (1.28)      4.1222 (1.0)      0.1382 (2.72)        31;34  236.9732 (1.0)         225           1
test_read_interleaved_data_channel        4.0924 (1.06)      27.9398 (3.76)      4.3430 (1.03)      1.5822 (5.70)      4.2263 (1.03)     0.0509 (1.0)          1;13  230.2550 (0.97)        225           1
test_stream_contiguous_data_channel       5.2963 (1.37)       8.8400 (1.19)      6.3609 (1.51)      0.5370 (1.94)      6.2584 (1.52)     0.2639 (5.18)          4;5  157.2094 (0.66)         34           1
test_read_contiguous_data_channel         6.1760 (1.60)       7.5365 (1.01)      6.4648 (1.53)      0.2774 (1.0)       6.3358 (1.54)     0.1867 (3.67)        17;16  154.6826 (0.65)         73           1
test_index_contiguous_data_channel       54.0179 (13.96)     94.0857 (12.66)    59.6456 (14.13)     9.0243 (32.53)    57.7451 (14.01)    2.2900 (44.99)         1;1   16.7657 (0.07)         17           1
test_index_interleaved_data_channel      77.0055 (19.90)    202.4484 (27.24)    96.0834 (22.77)    44.7370 (161.28)   77.5762 (18.82)    1.9825 (38.95)         2;2   10.4076 (0.04)         13           1
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

------------------------------------------------------------------------------ benchmark 'read-all-data': 2 tests -----------------------------------------------------------------------------
Name (time in ms)                  Min                Max               Mean            StdDev             Median               IQR            Outliers       OPS            Rounds  Iterations
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_read_interleaved_data      4.8834 (1.0)      31.0659 (1.92)      5.8230 (1.0)      2.0110 (8.77)      5.5595 (1.0)      0.4116 (1.63)          1;3  171.7329 (1.0)         163           1
test_read_contiguous_data      14.8667 (3.04)     16.1762 (1.0)      15.5787 (2.68)     0.2293 (1.0)      15.5264 (2.79)     0.2530 (1.0)          16;4   64.1900 (0.37)         63           1
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

-------------------------------------------- benchmark 'read-metadata': 1 tests -------------------------------------------
Name (time in ms)                     Min      Max     Mean  StdDev   Median     IQR  Outliers      OPS  Rounds  Iterations
---------------------------------------------------------------------------------------------------------------------------
test_complex_metadata_reading     10.0328  12.4886  10.6923  0.4325  10.7313  0.5734      17;2  93.5249      73           1
---------------------------------------------------------------------------------------------------------------------------

---------------------------------------- benchmark 'read-scaled-channel': 1 tests ---------------------------------------
Name (time in ms)                     Min     Max    Mean  StdDev  Median     IQR  Outliers       OPS  Rounds  Iterations
-------------------------------------------------------------------------------------------------------------------------
test_stream_scaled_data_chunks     8.9770  9.8324  9.1738  0.2024  9.1192  0.1096       2;2  109.0062      16           1
-------------------------------------------------------------------------------------------------------------------------

---------------------------------------------------------------------------------------- benchmark 'read-timestamp-data': 2 tests ---------------------------------------------------------------------------------------
Name (time in us)                               Min                    Max                  Mean              StdDev                Median                IQR            Outliers       OPS            Rounds  Iterations
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_read_interleaved_timestamp_data       791.2000 (1.0)      13,242.0960 (4.31)     1,005.5355 (1.0)      446.7277 (2.47)       977.2670 (1.0)      37.6770 (1.46)        11;93  994.4950 (1.0)         797           1
test_read_timestamp_data                 1,284.9540 (1.62)      3,074.1750 (1.0)      1,374.7622 (1.37)     180.5710 (1.0)      1,324.8410 (1.36)     25.8200 (1.0)         20;28  727.3985 (0.73)        248           1
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

[... cut ...]
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Legend:
  Outliers: 1 Standard Deviation from Mean; 1.5 IQR (InterQuartile Range) from 1st Quartile and 3rd Quartile.
  OPS: Operations Per Second, computed as 1 / Mean
=================================================================== 4 failed, 528 passed, 18 warnings in 94.61s (0:01:34) ===================================================================
*** Error code 1
```

Version: 1.9.0
Python-3.9
FreeBSD 14.0

adamreeve, 2024-04-08:
Hi @yurivict. I might be missing something, but I don't see anything about which tests failed in the output you've provided. The "invalid value encountered in reciprocal" warnings are concerning but I can't tell whether these are related to any test failures. Is there any extra output showing which tests failed? It might help to run pytest with the `-v` option to get more verbose output.

adamreeve, 2024-04-08:
Can you also please specify which version of numpy you are using

adamreeve, 2024-11-19:
Closing this as I can't reproduce it and would need more details

closed 2024-11-19 by adamreeve