adamreeve/npTDMS #331: The length of the channel data different from the length got from origin

MintBlue, 2024-06-17:
I hava encountered a strange problem,the data can be read with no error.But the length is 20240,and if you read the tdms file with originlab or flexpro,the length is 15240.
I don't know what made this happen,is it because I did not configure it correctly.I just use the code below
```python
with TdmsFile.open(file_path) as tdms_file:
     group = tdms_file[group_name]
     channel = group[column_name]
     column_data = channel[:]
```
[adq.zip](https://github.com/user-attachments/files/15861479/adq.zip)

MintBlue, 2024-06-19:
when segment_incomplete is True,can I drop the last segment data?

closed 2024-06-19 by MintBlue

adamreeve, 2024-06-24:
Hi @MintBlue 

In your file the "Next segment offset" property for the last segment is set to -1, which can sometimes indicate that the last segment wasn't completely written, but this is documented as a valid state in the TDMS format documentation (https://www.ni.com/en/support/documentation/supplemental/07/tdms-file-format-internal-structure.html).

I think for this file the last 5000 values seem to be legitimate data but it's hard to give a general answer for when the last segment of data is trustworthy if you see this warning, and whether you want to keep that data or not will depend on your specific use case and how much you care about reading all possible data.