adamreeve/npTDMS #339: Dealing with data that come through as a stream of bytes

ohmycaptainnemo, 2024-11-04:
Hi,

I really like npTDMS and am aware that it already allows for processing the TDMS file in chunks. But I could not find any examples of how to use npTDMS when the data is not in form of a file.
For example, I have a situation in which a large TDMS file comes through as chunks of bytes. Unfortunately, this means that even though reading the first chunk of bytes works as expected, the rest of the chunks result in the error:

```
ValueError: File should either start with 'b`TDSh`' or 'b`TDSm
```
or warnings such as:

```
[nptdms.reader WARNING] Last segment of file has less data than expected, will attempt to read to the end of the file
```

```
[nptdms.tdms_segment WARNING] Data size 3705 is not a multiple of the chunk size 7232. Will attempt to read last chunk
```

Which makes sense. Furthermore, in my case, I cannot read the large TDMS file into the memory and then wrap it in a `BytesIO` object as I do not have large memory at my disposal. So the only way I can think of is to process the chunks of data as they come through.

I would appreciate any thoughts or ideas regarding how it would be possible and how to use npTDMS for this use case.

adamreeve, 2024-11-19:
Hi @ohmycaptainnemo 

The `TdmsFile.open` method can work with any file-like object, it doesn't have to be a `BytesIO` instance. But it does expect the file to be seekable, and first reads all the metadata for each TDMS chunk before you can read any data. We also try to get the size of the file by seeking to the end and asking for the file position: https://github.com/adamreeve/npTDMS/blob/d2cddeb3c254927d7ea202ad97f31b0246bf7cd6/nptdms/reader.py#L522-L527

Assuming you know the file size in bytes ahead of time, and it's OK to iterate over all the byte chunks twice (once for metadata, and then once to read the data required, maybe with `TdmsFile.data_chunks`), then you should be able to implement your own wrapper class that implements the `seek`, `tell`, `read` and `readinto` methods and manages iterating through the chunks of bytes. This could inherit from [io.RawIOBase](https://docs.python.org/3/library/io.html#io.RawIOBase).

ohmycaptainnemo, 2024-11-19:
> Assuming you know the file size in bytes ahead of time, and it's OK to iterate over all the byte chunks twice (once for metadata, and then once to read the data required, maybe with [...]

@adamreeve Thank you for getting back to me. I did end up going through the code and indeed what you suggest makes sense. 

I did try to see if I can rely on an `index` file as a "header" so to speak which could tell me the position of all segments, but looking at the code, it seems that the index file could also sometimes not agree with the underlying tdms file.

adamreeve, 2024-11-19:
I think the cases where a `tdms_index` file gets out of sync with the data are usually caused by the data file being overwritten without updating the index file. If you're careful to avoid that then using an index file could help.

It's not currently possible to open a file object and also use a `tdms_index` file though, the index files are only used when a file path is specified. But that could be changed if needed.

adamreeve, 2024-11-19:
It should be possible to add a method to stream data in one pass, but this would be a fairly big change as npTDMS wasn't designed for that scenario, and it's not something I would have time to implement myself.

closed 2024-11-19 by ohmycaptainnemo