adamreeve/npTDMS #345: Issue with bool8 in numpy packages

Xinenomine, 2025-05-12:
Hello, 

I have dependency issue while working with npTDMS. Indeed, it clashes with the definition of bool8 that is depreciated since version numpy 1.24 with other packages. Unfortunately, I need to use recent version of numpy as I am using lmfit 1.3.3 [https://lmfit.github.io/lmfit-py/](https://lmfit.github.io/lmfit-py/) that requires numpy > 1.24 at least. 

My question is : are you planning to upgrade numpy to version 1.24 at least and get rid of depreciated bool8 numpy definition in the future ? 

Thank you very much, 
Have a great day, 
Best, 
Xinenomine

adamreeve, 2025-05-12:
Hi @Xinenomine 

npTDMS is tested against the latest numpy version and doesn't contain any references to bool8, so I don't understand what the issue is. Can you please provide more details of your problem, like an error message?

adamreeve, 2025-05-12:
It looks like this was fixed in #291, and version 1.6.1. Maybe you're using an old version of npTDMS? The latest version is 1.10.1.

Xinenomine, 2025-05-13:
Hello Adam, 
Thanks for your timely answer. I appreciate it. Maybe some context is needed: 
- I am trying to export a WASM-HTML sandbox app using Marimo python ( [https://docs.marimo.io/guides/exporting/#export-to-wasm-powered-html](https://docs.marimo.io/guides/exporting/#export-to-wasm-powered-html) . When doing so, I have this error (and only when trying to deploy the app, in editor marimo mode, there is no issue).
This is the error I get: 
```
Traceback (most recent call last):
  File "/lib/python3.12/site-packages/marimo/_runtime/executor.py", line 122, in execute_cell
    exec(cell.body, glbls)
  File "/tmp/marimo_42/__marimo__cell_Hbol_.py", line 2, in <module>
    from nptdms import TdmsFile
  File "/lib/python3.12/site-packages/nptdms/__init__.py", line 10, in <module>
    from .tdms import TdmsFile
  File "/lib/python3.12/site-packages/nptdms/tdms.py", line 6, in <module>
    from nptdms.tdms_segment import TdmsSegment
  File "/lib/python3.12/site-packages/nptdms/tdms_segment.py", line 8, in <module>
    from nptdms import types
  File "/lib/python3.12/site-packages/nptdms/types.py", line 166, in <module>
    @tds_data_type(0x21, np.bool8)
                         ^^^^^^^^
  File "/lib/python3.12/site-packages/numpy/__init__.py", line 410, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'bool8'. Did you mean: 'bool'?

```
It seems that it is linked to nptdms package (for my first order investigation) but I might be wrong. 
Thanks again for your help,
Best, 
Xinenomine

adamreeve, 2025-05-13:
That code is not present in the latest version of npTDMS, please try upgrading your npTDMS version.

closed 2025-05-15 by Xinenomine