adamreeve/npTDMS #346: TdmsWriter.defragment crashes when the input contains a channel with no data

auxym, 2025-06-13:
Here is the traceback:

```
Traceback (most recent call last):
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\scripts\tdmsdefrag.py", line 95, in <module>
    main()
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\scripts\tdmsdefrag.py", line 86, in main
    nptdms.TdmsWriter.defragment(args.input, args.output)
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\.venv\Lib\site-packages\nptdms\writer.py", line 51, in defragment
    channel.read_data(scaled=False),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\.venv\Lib\site-packages\nptdms\tdms.py", line 659, in read_data
    raw_data = self._read_channel_data(offset, length)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\.venv\Lib\site-packages\nptdms\tdms.py", line 866, in _read_channel_data
    for chunk in self._reader.read_raw_data_for_channel(self.path, offset, length):
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\.venv\Lib\site-packages\nptdms\reader.py", line 163, in read_raw_data_for_channel
    self._ensure_open()
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\.venv\Lib\site-packages\nptdms\reader.py", line 434, in _ensure_open
    raise RuntimeError(
RuntimeError: Cannot read data after the underlying TDMS reader is closed
```

This seems because when we get here: https://github.com/adamreeve/npTDMS/blob/8c8c44f7027c3971b61e66dc6ad5b078834439da/nptdms/tdms.py#L658 the attribute `raw_data` is `None`, which leads to attempting to read the data by calling the reader, which was closed at the end of TdmsFile's `__init__` (https://github.com/adamreeve/npTDMS/blob/8c8c44f7027c3971b61e66dc6ad5b078834439da/nptdms/tdms.py#L142).

I think the best solution would be to set `_raw_data` to some sort of empty array when it gets read during `__init__` when the channel contains no data, instead of `None`, that way we won't attempt to re-read it after the file was closed. I think this happens here: https://github.com/adamreeve/npTDMS/blob/8c8c44f7027c3971b61e66dc6ad5b078834439da/nptdms/tdms.py#L358

Alternatively, a quick patch could be to pass `keep_open=True` when opening the input TDMS file.

auxym, 2025-06-13:
For what it's worth, I tried the second option (keep the input file open), but it creates another issue:

```
Traceback (most recent call last):
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\scripts\tdmsdefrag.py", line 96, in <module>
    main()
  File "C:\Users\thef2102\source\mrfsplus_code\data analysis\scripts\tdmsdefrag.py", line 87, in main
    nptdms.TdmsWriter.defragment(input_file, args.output)
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 49, in defragment
    new_file.write_segment([ChannelObject(
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 151, in write_segment
    segment.write(self._file)
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 193, in write
    leadin = self.leadin(toc, metadata_size)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 242, in leadin
    next_segment_offset = metadata_size + self._data_size()
                                          ^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 253, in _data_size
    data_size += object_data_size(obj.data_type, obj.data)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\thef2102\source\npTDMS\nptdms\writer.py", line 451, in object_data_size
    return data_type.size * len(data_values)
           ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'NoneType' and 'int'
```

adamreeve, 2025-06-15:
Thanks for the bug report @auxym.

> I think the best solution would be to set _raw_data to some sort of empty array when it gets read during __init__ when the channel contains no data, instead of None, that way we won't attempt to re-read it after the file was closed

Yes this sounds like the right approach to me. Do you want to make a PR to fix this? Otherwise I'll try to fix this soon.

PR #348 (2025-07-25): Support reading and writing empty and/or void-type data segments
Fixes #346 

Add two tests:
* Read a TDMS file containing an empty channel (no data) that has no data type also
* Write and read back a segment with the void data type and no data

When reading a channel without data and without datatype, the returned value will be an empty numpy array of type `void`.

Void-type segments to be written must be empty (0 length data), otherwise an exception will be raised.
closed 2025-07-25 by adamreeve